evercam/ex_nvr #218: /footage endpoint to be able to specify resolution and format

marcoherbst, 2023-10-19:
This came from a conversation with a client who downloads 3min clips for manual review. For them, 1080p or less would be preferred.

I also think that when we start recording in h265, that footages would be by default delivered in h265 and it seems to me that a great NVR would offer the option to either "default" or specify a codec.

FerielKHA, 2023-11-02:
@marcoherbst what should be the format to allow ? This is the list that I've found: 

MP4
AVI 
MKV 
MOV 
WMV 
FLV 
WebM
3GP 
OGG 
VOB 
DivX 
H.264
H.265 (the DEFAULT one)

I think if we have:
MP4 (generally used when we download)
H.265 (the DEFAULT one)
It will be good. 

What about resolution. Should we display a list, and then the user could select one, and start the download ? 

[image]

marcoherbst, 2023-11-02:
Resolution: Yes, your idea is perfect. A list of common choices.
Format: I think we're only talking about h264 and h265, but I'd say yes to anything that membrane framework offers ... the more the better. 
NB. I think some of those other items are containers. I dont know much about this subject but i think .mp4 is perfectly ok for all use cases until I learn different.

FerielKHA, 2023-11-07:
NB for me. The snapshot to download corresponds to the moment when the user clicks the button to download it.

oussamabonnor1, 2026-10-01:
Closing for now

closed 2026-10-01 by oussamabonnor1