evercam/ex_nvr #220: API Docs published. Include a section on time.

marcoherbst, 2023-10-21:
1. I'd like to see the docs online somewhere independent of a running instance. Maybe on Github Pages?
https://github.com/peter-evans/swagger-github-pages
https://pguso.medium.com/host-openapi-documentation-with-github-pages-ec16f75e9762

2. Something worthy of inclusion in documentation would be an explainer on how we handle time, ntp, etc.  Points that have come up so far include:

- All times are in UTC.
- Get the camera timezone from the devices endpoint: https://wg2.evercam.io:21601/swagger/index-c435050a755f3ba71a4df97e599b71c9.html?vsn=d#/Device/getDevice 
- Default NTP server is: xxx

gBillal, 2023-10-21:
1. OK. I think github pages make sense for this use case
2. There's nothing to mention about that, using UTC is an implementation detail and used to save date times in database, the timezone is only used for presentation. For the endpoints, they do accept any datetime with timezone (not only UTC)
3. Not related. NTP config is a device/OS configuration and not a config of this software, this info should live in the devops repo.

marcoherbst, 2023-10-21:
Ok. 
On the NTP server, I'm fine with that position for now, but I believe this is the VMS (app) v NVR (os) debate. I believe the destiny of ex_nvr is to be implemeneted as a full system image and will include functionality for all of the OS functions that matter to a VMS, including HDD management, time settings and more. (as you would find on an NVR).

PR #242 (2023-10-31): Deploy documentation to github pages
resolves #220

I had to copy the swagger ui files and also openapi.yaml because symbolic links doesnt work with github pages.
A new version of the documentation will be published each time a tag is created.

The `gh-pages` branch should not be deleted when we merge this PR, it's used in the workflow.

link to documentation on github pages: https://evercam.github.io/ex_nvr/
closed 2023-10-31 by gBillal