evercam/ex_nvr #232: Deplay before video plays when clicking on the timeline

FerielKHA, 2023-10-25:
Using v0.5 
on https://dublin.evercam.io:8037/dashboard 
when I click along the timeline the image arrives quickly, but I have a delay before the video plays. 

See the behavior here --> https://www.screenpresso.com/fr/cloud/Ba3tf/

colttylerwilson, 2023-10-25:
i-frame is likely set to 1 in 50 on most Hik cameras we have, and 1 in16 on most Milesight cameras we have by default.

Might look for differences in CBR vs VBR in this case.

"The concept of Constant Bit Rate (CBR) stands on the principle that the bitrate level will be fixed regardless of other settings. Therefore, if we increase GOP value and extend the gap between I-frames then under CBR we cannot enjoy the benefits of optimal bandwidth consumption. Therefore GOP should not be used in combination with CBR. GOP should be used under Variable Bit Rate (VBR) mode instead."

https://www2.acti.com/support_old/Package/%7B8FA0B84F-A284-4AA2-BE94-538E4692ACD.PDF

marcoherbst, 2023-11-09:
It is quite possible that it doesn't make sense for this to be solved by ExNVR and instead, as @colttylerwilson is suggesting, this is simply a factor of how the camera stream is configured.  In this case, I'd like to consider the solution from a different perspective, maybe what we'd like to do is to include in our ExNVR documentation some words along the lines of "How to improve the seek performance of your recordings?".

This is a scenario where making it possible for a user to see via an info (i) button what the properties of the stream are would be interesting.

gBillal, 2024-01-05:
Fixed on #297

closed 2024-01-05 by gBillal