evercam/ex_nvr #241: Remove / Delete a device

marcoherbst, 2023-10-30:
[image]

I think there's a "Delete" option missing here.

FerielKHA, 2023-11-03:
@marcoherbst in which section I'll find this ? 
What's the impact of this action ? If the device is used ? Is it irreversible ?

marcoherbst, 2023-11-09:
Great question. 

My suggestion is that if we delete the camera, then the recordings are also deleted. Another approach would be some form of soft-delete or recycle bin, but for now, I would propose delete = delete.

I don't have absolute opinions on this, if someone wants to spec a more detailed answer, that's fine, especially if it references what other applications do.

closed 2023-12-21 by gBillal