evercam/ex_nvr #258: The Calendar Popup needs an "OK" button

marcoherbst, 2023-11-20:
From this view:
[image]

I don't know how to accept the chosen date/time.

Spotted by Waqas

gBillal, 2023-11-20:
This is the default calendar and we don't have a way to update it. Just select the date a click away and it'll be set (don't click outside the modal)

marcoherbst, 2023-11-20:
Could I get a second opinion on this? It feels wrong. 
I'd like to reopen this and tag @KamelBadji  
If this really is the normal way for a calendar to behave then I'm happy for Kamel to close it.

gBillal, 2023-11-20:
A button is unnecessary in this component because you have many ways of updating the calendar.

- You can type directly in the component without even opening the dialog (and use left/right keyboard keys to navigate)
- Click on enter will close the dialog also
- Click on the calendar button when you finish selecting a date and it'll also close

marcoherbst, 2023-11-20:
I have learned not to debate longer than 3 replies with Billal. Closing :)

closed 2023-11-20 by marcoherbst