evercam/ex_nvr #337: BIF: Inconsistent aspect ratio between BIF thumbnails and the HLS stream

halimb, 2024-01-25:
### Description

After the changes introduced by https://github.com/evercam/ex_nvr/pull/314 , BIF thumbnails have a different aspect in comparison with the HLS stream.

### Example
| Before [v0.8.0](https://github.com/evercam/ex_nvr/releases/tag/v0.8.0)| After [v0.8.0](https://github.com/evercam/ex_nvr/releases/tag/v0.8.0)|
| ------ | ----- |
|[image]| [image] |

### Desired outcome
Conserve the same aspect ratio as the main stream.

jakew009, 2024-01-28:
It's because the bif files are generated from the sub stream but the recording must be playing the main stream.

[image]

[image]

If the sub stream was changed to 640x360 the problem would go away

[image]

marcoherbst, 2024-02-25:
@halimb as Jake says, I dont think this is a big or an issue, it's just the way things are. 
@FerielKHA you might explain this in the wiki in case the question raises its head again and then close this issue.

halimb, 2024-02-25:
ok, in the meantime, I'll force the aspect ratio in the frontend to be the same as the main stream (different FoV, but not noticeable for the usecase)

marcoherbst, 2024-02-28:
I dont think this is an issue for the software to fix. In Evercam we will solve this by forcing the bif to be the same aspect ratio as the stream playing. In ExNVR we dont use the bif, so it's not a subject. Billal, reopen if you disagree.

closed 2024-02-28 by marcoherbst