evercam/ex_nvr #360: Download Progress/Acknowledgement Queue?

FerielKHA, 2024-02-07:
Goal:
To have a clear indicator that download is going to be processed and a time estimation (if possible).

gBillal, 2024-02-07:
???

FerielKHA, 2024-02-07:
When we click on download a record, nothing indicates that is in progress. We would like to have a message here.

gBillal, 2024-02-07:
There's a progress bar on the browser, why we should implement it ourselves ?

FerielKHA, 2024-02-13:
Just retested with @gBillal , I've tried to download 4h recording nothing happens (the error is not displayed)

FerielKHA, 2024-02-14:
@Mebareksaf a video that you asked explaining the issue 

https://github.com/evercam/ex_nvr/assets/51049639/1c5297cd-f0ea-45b1-80d4-4a96bff9c79a

Mebareksaf, 2024-02-14:
> @Mebareksaf a video that you asked explaining the issue 
> 
> 
> https://github.com/evercam/ex_nvr/assets/51049639/1c5297cd-f0ea-45b1-80d4-4a96bff9c79a
> 
> 

@FerielKHA Thank you so much, that's much better

Mebareksaf, 2024-02-14:
@gBillal i just noticed that this is a deployed ExNVR, does it use the version where we fixed the previous errors? (disconnected socket, end_date validation correction)

Mebareksaf, 2024-02-14:
@FerielKHA @gBillal the V0.9.1 does not contain the latest changes and fixes

gBillal, 2024-02-14:
@Mebareksaf @FerielKHA already fixed

closed 2024-02-14 by gBillal