evercam/ex_nvr #368: Limit viewers to one viewer of video per device

FerielKHA, 2024-02-12:
We choose to limit viewers to one viewer of video per device (or 2 or 3… whatever is possible)

> This means ExNVR returning a ( 503? ) and the Frontend giving a nice message.

jakew009, 2024-02-15:
What problem are we trying to solve with this issue? What is the bottleneck?

marcoherbst, 2024-02-15:
This:
https://github.com/evercam/ex_nvr/discussions/377

We like to record in h265  4k for as long as possible.

This means converting h265 to h264 on demand.

The first limiting factor is likely to be the Pi CPU ... with our current settings we can only do one stream at a time.

jakew009, 2024-02-15:
@marcoherbst why do we need to convert it to h.264? That's never going to work very well imo, the RPi just isn't powerful enough to transcode such a huge stream.

Most browses can play back H.265 video natively now. It doesn't work for WebRTC but it does for HLS. So why not just play the H.265 natively in the browser?

I don't know if you use hls.js but they've just merged this as well https://github.com/video-dev/hls.js/pull/5847 which helps Giraffe as we use .ts containers and this was a blocker previously.

But failing that why not get MediaMTX to convert from H.265 to H.264 in the cloud?

gBillal, 2024-02-16:
Hi @jakew009 
As per my research, the browsers (chrome/firefox) doesn't support H265. Did you have any luck playing H265 natively on this browsers ? Any resources or articles ?

> But failing that why not get MediaMTX to convert from H.265 to H.264 in the cloud?

We have bandwidth limits. So we want the bitrate to be smaller. (we'll scale down the video to 1080p or 720p)

jakew009, 2024-02-16:
@gBillal can you play this? https://hlsjs.video-dev.org/demo/?src=https%3A%2F%2Fbitmovin-a.akamaihd.net%2Fcontent%2Fdataset%2Fmulti-codec%2Fhevc%2Fstream_fmp4.m3u8&demoConfig=eyJlbmFibGVTdHJlYW1pbmciOnRydWUsImF1dG9SZWNvdmVyRXJyb3IiOnRydWUsInN0b3BPblN0YWxsIjpmYWxzZSwiZHVtcGZNUDQiOmZhbHNlLCJsZXZlbENhcHBpbmciOi0xLCJsaW1pdE1ldHJpY3MiOi0xfQ==

marcoherbst, 2024-02-17:
I think that given the recent discussion where, for our own use case, we plan to combine h264 recording with cloud multiplexing, this issue is less of a priority.

We may well still want to build in limits & appropriate messages, but I think that would be best handled with a separate GH issue when we have a better understanding of those limits.

closed 2024-02-17 by marcoherbst

devoldemar, 2024-02-18:
> As per my research, the browsers (chrome/firefox) doesn't support H265

Chrome supports HEVC if hardware acceleration is on in browser's Settings and graphics card can do the job.
> @gBillal can you play this? [...]

Chrome rejects codec `hvc1.1.c.L120.90` declared in the playlist. May be a bug or some compatibility requirements for codec name. But the video itself can be played. The way to check HEVC is supported is to open DevTools (F12), type in Console the following expression, and see if it will return `true`:
`MediaSource.isTypeSupported('video/mp4;codecs=hvc1.1.6.L120.90');`

gBillal, 2024-02-18:
> Chrome supports HEVC if hardware acceleration is on in browser's Settings and graphics card can do the job.

Hardware acceleration is enabled on my browser and my machine supports VA-API (tried with ffmpeg). However Hevc doesn't work. The reason perhaps is linked to my machine having AMD, I'll try it again on an intel machine

devoldemar, 2024-05-19:
> Hardware acceleration is enabled on my browser and my machine supports VA-API (tried with ffmpeg). However Hevc doesn't work. The reason perhaps is linked to my machine having AMD, I'll try it again [...]

If someone is interested, [here's](https://devoldemar.github.io/hevc) a simple webpage to test browser's general support of H.265 against popular resolutions and FPS (Main 1K ~ 4K should be considered for video surveillance). And instructions how to possibly turn on this support .