evercam/ex_nvr #369: Catchall issue for lots of small ideas (Placeholder)

marcoherbst, 2024-02-12:
NB Some of these items may not be ExNVR related, but should live upstream (Evercam) ... consider this a placeholder for ideas which can be cleaned up later. I felt they were all too small to deserve their own issue but needed to live somewhere.

- [ ] LiveDashboard in an IFRAME rather than as a new Tab.
- [ ] https://github.com/acalejos/merquery as a way to play with APIs on the camera. 
- [ ] Livebook embedded or open in new tab. 
- [ ] /live-dashboard/ecto_stats  can we get something useful from this?
- [ ] Rename API Documentation to API Docs
- [ ] Don't use bold font in the left nav
- [ ] Make the video player full width

marcoherbst, 2024-02-15:
@Mebareksaf you can take any of these relatively small items that you feel able to do. Discussed already with Billal. The more the merrier !

the-prksh, 2024-08-09:
Hey 👋 ,

I am currently integrating `livebook ~> 0.13.3` and I have encountered a problem with dependency resolution. Livebook  depends on `phoenix_live_view ~> 1.0.0-rc.5`, while another project dependency, `flop_phoenix`, depends on `phoenix_live_view ~> 0.20.4`.

I am struggling to find a way to resolve this conflict. Can anyone provide some insight on how to address this issue?

Here is the output of `mix` deps resolution:

```shell
. . .
Because your app depends on livebook ~> 0.13.2 which depends on phoenix_live_view 1.0.0-rc.5, phoenix_live_view 1.0.0-rc.5 is required.
So, because your app depends on phoenix_live_view ~> 0.20.0, version solving failed.
```

```shell
. . .
Because flop_phoenix >= 0.22.7 depends on phoenix_live_view ~> 0.20.4 and flop_phoenix >= 0.21.2 and < 0.22.7 depends on phoenix_live_view ~> 0.19.0 or ~> 0.20.0, flop_phoenix >= 0.21.2 requires phoenix_live_view ~> 0.19.0 or ~> 0.20.0.
And because your app depends on flop_phoenix ~> 0.22, phoenix_live_view ~> 0.19.0 or ~> 0.20.0 is required.
So, because your app depends on phoenix_live_view ~> 1.0.0-rc.5, version solving failed.
** (Mix) Hex dependency resolution failed
```

gBillal, 2024-08-10:
I don't think including `livebook` as a dependency is a good idea. It's better to run a standalone livebook, creating the livebooks and connecting it to a running instance of `ex_nvr` to do the actual job.

If you still want to include it directly, either:  
* Upgrade the other dependencies to meet the requirements.
* Or find an older version of `livebook` that doesn't cause conflicts with the actual dependencies.

the-prksh, 2024-08-14:
Running a standalone node locally and connecting it to a running instance of `ex_nvr` on a Raspberry Pi is not possible because `ex_nvr` runs on Elixir 1.15, while for Livebook the node must run Erlang/OTP 26 and Elixir 1.16.3 (or later). Here's a screenshot for reference: 
[image]

gBillal, 2024-08-14:
Unfortunately we cannot upgrade elixir to the latest version on `ex_nvr` for legacy reasons. However you can downgrade `livebook` to meet the requirements.

the-prksh, 2024-08-15:
May I know the legacy reasons? 

I have upgraded the Elixir version in the Dockerfile to `hexpm/elixir:1.17.2-erlang-27.0-alpine-3.18.7` and everything seems to be working fine.

the-prksh, 2024-08-16:
I have deployed and checked in Raspberry Pi as well, and it is working fine there as well. Would you like to have a look?

the-prksh, 2024-08-16:
When it comes to the Standalone livebook node in local machine and connecting runtime with edge device, 

It won't function on edge devices because the device doesn't have `mix` installed. Due to this, we are unable to install dependencies dynamically using `Mix.install/1`. 

You might be curious about how `nerves_livebook` addresses this issue. They tackle it by including all the `livebook` related dependencies as part of the release image and tweaking the `Mix.install/1` functionality. Here is the code for that - https://github.com/nerves-livebook/nerves_livebook/blob/main/lib/nerves_livebook/mix_install.ex

[image]

marcoherbst, 2024-08-17:
Do I understand correctly, that to use Livebook on a Pi, any dependencies would have to have been included in the firmware build? 

Is that true only if we were using nerves? (which at the moment we're not), or is this some other limitation related to a Pi that even with a regular OS (Debian, PiOS), Mix cannot be used?

the-prksh, 2024-08-19:
That is true in case of integrating using 1. Attached Node or 2. Embedded Mode but partially true in case of integrating using `Standalone Node`.

the-prksh, 2024-08-19:
It is true when integrating using an Attached Node or Embedded Mode, but only partially true when integrating using a Standalone Node.

It has nothing to do with nerves. The Pi OS will be trimmed compared to a regular OS. In our case, the Production release of `ex_nvr` will not contain Mix and other development dependencies.

the-prksh, 2024-08-19:
I prefer using `embedded mode`. Nerves has used this approach to integrate Livebook. All dependencies will be included in the ex_nvr release, so there's no need to install anything on your local machine. Just use the link to open Livebook.

gBillal, 2024-08-19:
Hi
The legacy reasons I'm talking about is that we're using devices (rasberry pi) with old debian versions. If we use Elixir 1.17.x to create the releases it'll include dependencies that'll not work on old versions (however an old version works fine on newer versions).

For livebook with attached mode, you cannot install dependencies, however you can do the following:

- Include the dependencies you need in exnvr itself and then use attached mode in livebook
- Or use smart cell (remote execution) to only get the data you need from exnvr and then use the livebook runtime to process that data. In this case you can use `System.cmd` or `System.shell` to execute commands on the remote machine (e.g. turn on/off wifi)

the-prksh, 2024-08-19:
Hello @gBillal ,

Are you saying that some of the production kits contain old Debian versions that may cause issues? 

We are not changing the Debian version, only upgrading the Elixir/Erlang version. So, I don't think there would be any issue in that case. Is there any way we can test that?

Also, in `attached mode`, atleast elixir >= 1.16 is required.

the-prksh, 2024-08-19:
> * Or use smart cell (remote execution) to only get the data you need from exnvr and then use the livebook runtime to process that data. In this case you can use `System.cmd` or `System.shell` to execute commands on the remote machine (e.g. turn on/off wifi)

This will be possible in "Standalone Mode", but for that, a livebook has to be set up on the laptop. @marcoherbst explained that he expects to integrate the livebook into the ex_nvr dashboard. This will not be possible if we go with `Standalone Mode`.

gBillal, 2024-08-19:
> We are not changing the Debian version, only upgrading the Elixir/Erlang version. So, I don't think there would be any issue in that case. Is there any way we can test that?

Elixir 1.17 docker images are all based on the newer debian version and if that image is used to build arm64 releases, the external dependencies (like `ex_dtls`, `libsrtp2`, ..) will depend on newer version  of GLIBC which will not be loaded on old version of debian. I already tested this.

Anyway you can work on a new branch, upgrade elixir and any dependencies you need and create a release from there. Unfortunately we cannot upgrade right now.

the-prksh, 2024-08-19:
@gBillal  Is it possible to upgrade Elixir to version `1.16` in that case?

gBillal, 2024-08-20:
As I said above, this work won't make it to the main branch. So you're free to work on a branch, upgrade to any version and create a release from there where we can test it on some devices.

we'll upgrade Elixir once we're no longer using old raspbian version on our side.

marcoherbst, 2024-10-26:
Related: https://elixirforum.com/t/til-accessing-a-nerves-device-from-livebook/67028