evercam/ex_nvr #407: ExNVR recordings with Hanwha cameras

smaude87, 2024-03-21:
There is a recording issue on the Hanwha branded cameras where the recordings are not continuous.
[image]

I have NX recording on the same server and there seems to be no interruptions in the continuous recordings.
[image]

Camera ID: auhan-ilvuo

Camera model: Hanwha PNO-8091R

gBillal, 2024-03-21:
Hi @smaude87 
Can you please try to free space on the device so we can disable the overwriting in the NVR. There's a bug there, so I cannot point the exact issue.
Either put this camera on a new device or delete old recordings from the hard drive

gBillal, 2024-03-21:
it's recording for a few hours now (the bug is in the delete old recordings), I'll try different config and investigate the issue more
[image]

smaude87, 2024-03-22:
@gBillal i had to return the camera today, we are shipping them to a Philippines project.  I have another camera on the way to keep the test rolling.  I suspect we will be using this brand of camera frequently in asia.

gBillal, 2024-03-22:
great @smaude87, keep me updated when the new camera is online.

closed 2024-03-22 by gBillal