evercam/ex_nvr #428: snapshots milisecond accuracy

Diaislam, 2024-05-10:
Description: 
the endpoint to request a snapshot from exnvr is not accurate with miliseconds 
camera exid newwa-dyrua
Example: 

image request newwa-dyrua_2024_04_19_10_58_05_400 , second: 5 , milisecond = 100
[image]
image request  newwa-dyrua_2024_04_19_10_58_05_700 second: 4 milisecond = 700 is the same exact image 
[image]

600 miliseconds difference but getting the same image

gBillal, 2024-05-10:
Hi @Diaislam 
I tried with this camera I get the following results
`2024-04-19T10:58:05.000Z`
[image]

`2024-04-19T10:58:05.500Z`
[image]

The two images are not the same. Are you sure that you provide the right timestamps ?

Diaislam, 2024-05-10:
[image]
yes correct but if you try the timestamps i sent milisecond 100 & milisecond 700 they are the same image ,
the 

if you try  2024-04-19T10:58:04.500 and 2024-04-19T10:58:05.000 they are also the same image, what i'm trying to say is in one second there are only 2 images mostly

gBillal, 2024-05-10:
Nope they're not the same, I tried with 100 millisecond and 300 millisecond and it gave me different images

closed 2024-05-10 by Diaislam