evercam/ex_nvr #484: Milesight camera not sending a picture

lawik, 2024-09-03:
I bought a fairly basic Milesight IP camera. It is set up on the network and I can get snapshots from it manually. It shows video.

ONVIF discovery in ex_nvr found it fine and seemed happy. But it never actually pulls any image from the device. Nothing of interest happens in the logs when hitting record or any of that.

Are there some good steps to try to figure out the issue or something obvious I must do? RTSP seems on by default.

gBillal, 2024-09-04:
Hi @lawik 
If the onvif discovery works fine, you can create a device from there and it should start pulling rtsp stream and record it.
Can you share some screenshots of what you did and the logs if possible.

lawik, 2024-09-04:
I created the device from ONVIF. That worked fine. It said it was recording. But nothing showed up in the camera view. No errors in log.

Will try to grab some more info next time I have it up.

gBillal, 2024-09-04:
is the camera using H265 codec  ? if so it may be the cause and your browser doesn't support hevc.
I'll wait for more details to debug the issue.

lawik, 2024-09-04:
I did not turn on h265. But I'll try to get more info. But I don't think there were any clear messages about problems or anything.

lawik, 2024-10-30:
Now I got an error trying to remove and add it again:

```
2024-10-30T13:28:12.415 [error] GenServer #PID<0.3484.0> terminating
** (KeyError) key :network_interface not found in: %{
  url: "http://192.168.3.120:80/onvif/device_service",
  date_time_settings: %{
    type: "Manual",
    timezone: "UTC-08:00:00",
    daylight_savings: true
  },
  device_information: %{
    model: "MS-C5363-PC",
    firmware_version: "45.7.0.79-r3",
    manufacturer: "Milesight Technology Co.,Ltd.",
    serial_number: "CM99V122132000242",
    hardware_id: "V1.2"
  },
  network_interfaces: nil,
  media_profiles: []
}
    (ex_nvr_web 0.15.2) lib/ex_nvr_web/live/onvif_discovery_live.ex:104: ExNVRWeb.OnvifDiscoveryLive.handle_event/3
    (phoenix_live_view 0.20.17) lib/phoenix_live_view/channel.ex:508: anonymous fn/3 in Phoenix.LiveView.Channel.view_handle_event/3
    (telemetry 1.3.0) /home/lawik/projects/ex_nvr/deps/telemetry/src/telemetry.erl:324: :telemetry.span/3
    (phoenix_live_view 0.20.17) lib/phoenix_live_view/channel.ex:260: Phoenix.LiveView.Channel.handle_info/2
    (stdlib 6.0.1) gen_server.erl:2173: :gen_server.try_handle_info/3
    (stdlib 6.0.1) gen_server.erl:2261: :gen_server.handle_msg/6
    (stdlib 6.0.1) proc_lib.erl:340: :proc_lib.wake_up/3
```

lawik, 2024-10-30:
[image]

lawik, 2024-10-30:
This was going through ONVIF Discovery.

lawik, 2024-10-30:
Formatting the SD Card made a network interface appear and it seemed happier to be added.

Still nothing coming up.

gBillal, 2024-10-30:
Hi @lawik,
If I remember correctly, you said that you're running the `master` branch and not using one of the releases. In this case most probably it's `run_pipelines` setting in `config.exs` or `dev.exs` is the issue. By default it is set to `false` to not run any video pipeline, you have to set it to `true`.

lawik, 2024-11-01:
Exactly right!

Flipping that makes it go :)

closed 2024-11-01 by lawik