evercam/ex_nvr #523: Onvif UI improvements

gBillal, 2024-12-12:
- [ ] Allow selection of stream profiles to use.
- [ ] Allow creating stream profiles for some cameras (e.g. `axis` and `hanwha`)
- [ ] Allow providing the network interface where to broadcast discovery messages.

xromansx, 2025-03-14:
Would be also good to have Onvif events monitoring (like record on motion event)

marcoherbst, 2025-03-15:
I thought that these docs were quite high quality reference, not just for onvif but for event handling of all sorts of manufacturers https://www.geutebrueck.com/g-help/g-core-ati/Content/IP-Camera%20Plugins/ONVIF%20IPC/Event%20Handling/Event%20Handling.htm?tocpath=IP-Camera%20Plugins%7CONVIF%7C_____5

@halimb 

I always get a bit lost in the onvif docs, but i think the one that's relevant is "Action Engine" 
https://www.onvif.org/specs/DocMap.html