evercam/ex_nvr #583: Rename "IP Address" to "Network Interface"

marcoherbst, 2025-02-21:
In Onvif discovery: 

[image]

I think this is more properly understood as a choice of Network Interfaces, not IP addresses.

gBillal, 2025-02-22:
Nope that's the IP address of your network interface (and it's the one used in the code). You can still have your network interfaces without an IP address and it'll not be listed here. I added just the name of the interface to make it easy to choose the correct IP address.

closed 2025-02-22 by gBillal

gBillal, 2025-02-22:
[image]

You can even have multiple IP addresses for the same interface. Perhaps I'll consider this use case.