evercam/ex_nvr #625: Support for Meanwell DRC-100 UPS

marcoherbst, 2025-04-09:
It has relays for AC Power and Low Battery.

[image]

[image]

This is how Chat GPT would do it:

## 🔌 Nerves Project: GPIO-Based Relay Monitoring with LiveView UI

This setup monitors **two GPIO input relays** on a Raspberry Pi 5 using Elixir/Nerves and displays their status in real-time via **Phoenix LiveView**.

---

### 🧱 Hardware Wiring

| Signal         | GPIO Pin (BCM) | Physical Pin |
|----------------|----------------|---------------|
| Low Battery    | GPIO 17        | Pin 11        |
| Mains AC       | GPIO 27        | Pin 13        |
| Ground         | —              | Pin 6         |

Each relay should be connected between the corresponding GPIO pin and **GND**, acting like a switch:
- When **closed**, the GPIO reads `LOW` (continuity detected).
- When **open**, the GPIO reads `HIGH` (no continuity).

Use internal pull-ups for each pin.

---

### ⚙️ GPIO Logic (Relay Monitor)

**File:** `lib/my_app/relay_monitor.ex`

```elixir
defmodule MyApp.RelayMonitor do
  use GenServer
  alias Circuits.GPIO

  @low_battery_pin 17   # BCM 17 (Pin 11)
  @mains_ac_pin 27      # BCM 27 (Pin 13)

  def start_link(_) do
    GenServer.start_link(__MODULE__, nil, name: __MODULE__)
  end

  def init(_) do
    {:ok, low_gpio} = GPIO.open(@low_battery_pin, :input, pull_mode: :pullup)
    {:ok, mains_gpio} = GPIO.open(@mains_ac_pin, :input, pull_mode: :pullup)

    GPIO.set_interrupts(low_gpio, :both)
    GPIO.set_interrupts(mains_gpio, :both)

    state = %{
      low_battery: GPIO.read(@low_battery_pin) == 0,
      mains_ac: GPIO.read(@mains_ac_pin) == 0,
      pins: %{
        @low_battery_pin => :low_battery,
        @mains_ac_pin => :mains_ac
      },
      gpios: %{
        @low_battery_pin => low_gpio,
        @mains_ac_pin => mains_gpio
      }
    }

    notify_clients(state)
    {:ok, state}
  end

  def handle_info({:circuits_gpio, pin, _ts, 0}, state) do
    key = state.pins[pin]
    new_state = Map.put(state, key, true)
    notify_clients(new_state)
    {:noreply, new_state}
  end

  def handle_info({:circuits_gpio, pin, _ts, 1}, state) do
    key = state.pins[pin]
    new_state = Map.put(state, key, false)
    notify_clients(new_state)
    {:noreply, new_state}
  end

  defp notify_clients(state) do
    Phoenix.PubSub.broadcast(MyApp.PubSub, "relay_status", {:relay_status, %{low_battery: state.low_battery, mains_ac: state.mains_ac}})
  end
end

gBillal, 2025-04-09:
@marcoherbst did you write the Elixir module ?

marcoherbst, 2025-04-09:
No no ... this was just a few mins of chat GPT.

"how to test circuit continuity on a pi" 
"can i have elixir circuits code"
"make the liveview UI"
etc

I have no clue if the results are meaningful, but I learned a little bit

gBillal, 2025-04-09:
almost correct 😄 
Do we have any kit with this setup ??

marcoherbst, 2025-04-09:
I'm afraid not, I didn't know the correct wiring until just now, but tmrw I will ask @FZECAM and Faisal to wire one up. 

They have several of the meanwell devices and are installing them like this:

[image]

On the left you can see the blue battery.

marcoherbst, 2025-04-09:
This is my understanding of what ChatGPT is suggesting.

@FZECAM 

[image]

gBillal, 2025-04-09:
I don't think it's that simple. What's the output of the relay when closed ? the GPIO pins works only with 3.3v with 6mA max per pin, otherwise we'll fry the rpi.

marcoherbst, 2025-04-09:
I think you're asking me what voltage the UPS provides over the relay? 

I think the answer is that the UPS does not provide any voltage, it is simply an on/off circuit breaker. 

The voltage comes from the Pi itself via the ground pin (6) which is 3.3v.

We are reading here only Open or Closed. High or Low. 

That's what I understand from the first diagram

[image]

Maybe I misunderstood your question.

gBillal, 2025-04-09:
Actually that's not how it works. The pins should be set as input, this means the relay will provide power to the pins, if there's power it means HIGH otherwise LOW. I read the documentation and the relay should powered separately, by looking at the schema providing the right voltage + the right resistance will make it work as expected. We should ask Faisal for this.

Read point 2

[image]

marcoherbst, 2025-04-09:
I am reading that and I think we might be both saying the same thing. 

In the picture above (From the UPS manual), the "external voltage source" would be provided by the Pi (pin 6, a ground pin, providing 3.3v).

This is what Google says:

[image]

and here's a simplified diagram:

[image]

Google does suggest adding a resister, but says it's optional.

gBillal, 2025-04-10:
Again Maco no, it cannot be done that way. A GPIO pin can be either used as input or output. If it's an output, it means we write to it from software side, you set it to HIGH it outputs 3.3v, you set it to low, it'll be 0v. As an input, we'll monitor the status of the pin, if from outside, there's voltage, it'll be HIGH otherwise LOW.
So the rpi is not the one providing power to the pins, but the relay should provide power to the pins and ExNVR will monitor those pins and react accordingly.

You should use the resistor if it's no included in the DRC, otherwise we'll fry the pins and damage the rpi if the current is too high.

gBillal, 2025-04-10:
Something that we can try is to use another pin as output to supply 3.3v to the relay. (I don't have enough knowledge on this, but I think it may work).

marcoherbst, 2025-04-10:
ok, what is your source of information? Mine is only google & chatgpt, but the way I read those screenshots it is telling me to create a circuit from a GND pin to a GPIO pin.

We both are agreeing that the GPIO pin will be reading HIGH/LOW, the only difference is we are discussing where the voltage is provided from. I am saying it is a GND pin and you are saying it should come from elsewhere.

We are also both agreeing about the resistor, just I am saying it is optional (but recommended) according to some of the results I saw.

This guy is doing something very similar. He is using a button press to simulate the UPS relay. 
https://www.youtube.com/watch?v=Ns0JdZrXwdk

[image]

marcoherbst, 2025-04-10:
> Something that we can try is to use another pin as output to supply 3.3v to the relay. (I don't have enough knowledge on this, but I think it may work).

"use another pin to supply..." ... YES. The Ground Pin. Pin 6. That's the idea. (or in the video , he uses pin 25 ... you can use any of the ground pins .... which have the 3.3v output.

gBillal, 2025-04-10:
@marcoherbst I hope I still remember some of my electronic classes.

[image]

Forget about the voltage source for now, I think we agree on our diagrams, the PIN8 in the schema will be wired to an input pin in the rpi, and the PIN7 to a ground pin. So if the relay is open, the pin will read HIGH and if closed, there'll be short circuit and the pin will read 0v.

The rpi pins only accepts 3.3v, so we can use another GPIO to supply power (not a ground one, because it cannot be used as output).

marcoherbst, 2025-04-10:
[GPIO_Continuity_Test_RPi5_with_Schematic.pdf](https://github.com/user-attachments/files/19694607/GPIO_Continuity_Test_RPi5_with_Schematic.pdf)

I dont have enough knowledge myself, but ChatGPT seems confident we can create a voltage between pins 6 and 11 by setting the pull-up resistor on pin 11 (GPIO).  

EIther way, as you say, the alternative is to use one of the 3.3v sources and, yes, resistors seem to be recommended (but I only saw the word recommended, not essential).

gBillal, 2025-04-11:
@marcoherbst you're overestimating ChatGPT for this use case. The pull-up and pull-down resistors have other use cases and the resistor is REQUIRED, we're building a kit to run for a long time, not a test to run for 1 hour.

marcoherbst, 2025-04-11:
Ok, then we're in agreement. 

My understanding is: 

1. We can test/develop without a resistor and for production use a resistor. (The behaviour will be the same, worst case we lose some Pis).
2. On every other aspect, we're in agreement. (Pin arrangement, High-Low etc.)

marcoherbst, 2025-04-12:
Just for curiosity at this stage, here is Gemini's answer, with resistor & using the 3.3v pin as a source:
https://docs.google.com/document/d/1MKCL2V3mWrPCuboBUc-y-uwskmfPuB44AdHIVYUPnrg/edit?tab=t.0

closed 2025-04-22 by gBillal