evercam/ex_nvr #673: Device details page

gBillal, 2025-05-02:
Currently we don't have any page that shows device details. We want to add a page that's accessible once we click on the device on the devices table.

The page details will be tab oriented and should include the following tabs as a starter:

- [ ] Details tab: should show information about the device including name, type, status, creation date, streams, ...etc. It should also include a small player or just an img to show video stream or a snapshot from the device.
- [ ] Recordings tab: shows the current recorded videos of the device in the timezone of the device + all the filters. In this tab we should also be able to download footage.
- [ ] Stats tab: this tab will show real time stats about the stream: bit rate, resolution, total bytes received, frame rate, ...etc.
- [ ] Settings tab: in this tab we'll be able to update/start or stop recording/deleting the device.
- [ ] Events tab: this tab will show only events related to the device.
- ... Other tabs will come later.

We should at least get the device details page as a starting point. Once this done we can delete the actions from the device list and the whole recordings tab.

@halimb @salah-walid do you think you can work on this ?

halimb, 2025-05-03:
sounds good to me.

@salah-walid unless you want to work on this, I can start working on it late week next week

salah-walid, 2025-05-07:
> sounds good to me.
> 
> @salah-walid unless you want to work on this, I can start working on it late week next week

@halimb no worries, you can take it