evercam/ex_nvr #776: Recording fails with valid RTSP URI on ExNVR (works in VLC)

colttylerwilson, 2025-08-19:
**Steps to repoduce**
1. Verified RTSP URI works in VLC:
rtsp://user:pass@192.168.1.69:554/h264Preview_01_main

[image]

2. Added same RTSP URI into ExNVR device setup (username + password provided).

[image]

3. Tried setting recording storage path to /root (SD card) and /root/media (external HDD).

[image]

4. Both attempts failed; recording does not start.

**Expected behavior**
ExNVR should be able to record the RTSP stream to the selected storage.

**Actual behavior**
Recording fails regardless of path selected.

**Notes**
Raspberry Pi and camera are on the same local network.
RTSP stream confirmed working via VLC.
Unsure if this is a storage permissions issue, config issue, or bug in ExNVR handling of storage paths.
Camera Model: Reolink E1 Pro
https://reolink.com/product/e1-pro/?srsltid=AfmBOor-6VcYcfVRHGjkLfp3JW8S_CBopqEGDPbUZJCzKGigNLvWAy5s

MosesMuiru, 2025-08-21:
> > <img alt="Image" width="1164" height="299" src="https://private-user-images.githubusercontent.com/93126287/480369201-40c8fd6d-304a-41fa-abce-d2bb545daded.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3ODc0NDQsIm5iZiI6MTc1NTc4NzE0NCwicGF0aCI6Ii85MzEyNjI4Ny80ODAzNjkyMDEtNDBjOGZkNmQtMzA0YS00MWZhLWFiY2UtZDJiYjU0NWRhZGVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDE0MzkwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNhZDQ5NTcwMjMzNWMwNDZkYzQwMmZhZDY3YjIwNmI1ODBhMGJkYjk2ZmViNTkzMTVjNzNkNTE1Y2QxZmY5NGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.HiluyzhbfXIJpzu9yR4ShRwkbV5JFEHzkVYZTVKo73I">
> > Hi [[@colttylerwilson](https://github.com/colttylerwilson)](https://github.com/colttylerwilson), try using onvif discovery
> 
> [@MosesMuiru](https://github.com/MosesMuiru) Thanks for popping in here. I tried Onvif discovery. On version 21 it was able to be scanned and found on local network but this version (0.23.0) seems to produce no results.
> 
> <img alt="Image" width="1511" height="358" src="https://private-user-images.githubusercontent.com/18061306/480556531-e59fdf95-f4ff-4521-ab7c-dabbbef03371.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTU3OTMwNTUsIm5iZiI6MTc1NTc5Mjc1NSwicGF0aCI6Ii8xODA2MTMwNi80ODA1NTY1MzEtZTU5ZmRmOTUtZjRmZi00NTIxLWFiN2MtZGFiYmJlZjAzMzcxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODIxVDE2MTIzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlMDBkN2U5NjBhMzllY2QzMzJiZDFmYmQ2YzBjYTJlMzZiMDExNTQ3NGFlMWI3OGNhMzk3MmZiYmRhZWFhNmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.QEVWW28ZWIUsFIqqV_pXp41hg1R7C-xPBG1muWTJLrA">

Maybe double check if onvif is enabled on your camera network settings.

colttylerwilson, 2025-08-21:
> Maybe double check if onvif is enabled on your camera network settings.

Yessir it is enabled @MosesMuiru and it was working on version 21.

Just seems so odd. It should pull an RTSP stream which is confirmed working on VLC, I'm not sure what else would change. If the RTSP works then it works.

[image]

gBillal, 2025-08-21:
@colttylerwilson the issue is that Reo-link cameras rtsp implementation is buggy. I'll try to add some edge cases to fix this, however it may take some time since that's an external lib that I need to patch.
The other solution is to use RTSP with UDP (I'll try this)

colttylerwilson, 2025-08-21:
Reolink E1 Pro
== Media Profiles ==
- [0] Profile000_MainStream (token: 000)
    RTSP URI: rtsp://192.168.1.69:554/
    Snapshot URI: http://192.168.1.69:80/cgi-bin/api.cgi?cmd=onvifSnapPic&channel=0
- [1] Profile001_SubStream (token: 001)
    RTSP URI: rtsp://192.168.1.69:554/h264Preview_01_sub
    Snapshot URI: http://192.168.1.69:80/cgi-bin/api.cgi?cmd=onvifSnapPic&channel=0

colttylerwilson, 2025-08-21:
Snapshot acquired using instead:

ffmpeg -rtsp_transport tcp -i "rtsp://user:pass@192.168.1.69:554/h264Preview_01_sub" -frames:v 1 snap.jpg

[image]

colttylerwilson, 2025-08-22:
Updated firmware on Reolink E1 Pro provided by Reolink Support.

Firmware file: IPC_566SD54MP.5112_2507081475.E1-Pro.4MP.WIFI7.PT.REOLINK.pak

colttylerwilson, 2025-08-22:
**Update (22 Aug 2025 – ~9:00–9:30 AM MX GT):**

Yesterday I adjusted the camera’s video settings (attached screenshot), but that did not resolve the issue; recordings were still not present.

However, starting this morning around 9:00–9:30 AM (Mexican General Time), recordings began appearing on the EXNVR. See attached screenshot of a sample recording.
	•	There is still a noticeable gap in the timeline where recordings are missing.
	•	Recording has resumed, but the stream is not yet stable; intermittent rather than continuous.
	•	I had received a firmware file from Reolink yesterday, and have just applied it now.

So, something clearly changed this morning (unclear what), and while the EXNVR is now capturing recordings again, the stability issue remains until proven otherwise with time.

Attachments: current video settings, sample recording screenshot.

[image]

[image]

closed 2025-09-16 by colttylerwilson